BCDA-APS/BITS #3: Basic instructions to run/test the queueserver from the queue monitor would help too; e.g.

prjemian, 2024-10-17:
Basic instructions to run/test the queueserver from the queue monitor would help too; e.g. 

- Connect to the host
- Change tab, open the environment
- Open plan editor, add sim plans to queue
- Start queue
The simulated plan run so fast that  it is hard to tell if they actually ran, need to look at the console (on the other tab). Not trivial for someone who has never used the queueserver.

_Originally posted by @rodolakis in https://github.com/BCDA-APS/bs_model_instrument/issues/2#issuecomment-2417771698_

prjemian, 2024-10-17:
The QS host interacts with `redis` at this network host:port: https://github.com/BCDA-APS/bs_model_instrument/blob/dda801d56acf454cc55b6bb96efdee4b4448eec4/qs/qs-config.yml#L13

QS clients do not need to know about this connection.  They interact only with the QS host.

prjemian, 2024-10-17:
[image]

Right.  If the QS host wanted to use the redis service which was running on a different workstation (say `example.server.org`), then change `qs-config.yml` as you said:

```yml
  redis_addr: example.server.org:6379 
```

prjemian, 2024-10-17:
> Basic instructions to run/test the queueserver ...

I'm sure we wrote these instructions in one of the BDP demo presentations.  I'll dig that up.

rodolakis, 2024-10-17:
I am happy to write those

> > Basic instructions to run/test the queueserver ...
> 
> I'm sure we wrote these instructions in one of the BDP demo presentations. I'll dig that up.

prjemian, 2024-10-17:
These instructions will need some translation into our current names.  They were used at the time of this presentation (2024-02-09).  For completeness, I'm copying the subset of slides.  Includes APS Data Management (which is an additional feature).

## general

[image]

[image]

[image]

To watch the QS Host console on your terminal, use this bash command: `qserver-console-monitor`.
Type ^C to exit this.

<details>

## when using data management

[image]

[image]

[image]

[image]

[image]

[image]

</details>

prjemian, 2024-10-17:
QS Host and QS clients are separate processes.

- QS Host process is started from a workstation with access to:
  - the `qs-config.yml` file
  - a `redis` service
  - the instrument package with the RE, devices, and plans
- Any QS client is started from a workstation with network access to the QS Host process

prjemian, 2024-10-17:
Your question involved the `qs-config.yml` file and port 6379 (port used by the redis service).    This configuration file is used when running `start-re-manager` (the program that actually operates the queueserver host process).

I'll look again at the `qs_host.sh` (which calls `start-re-manager`) to make sure that its comments and code agree with our documentation.  The shell script has some defense against certain beamline staff trying to start the QS host on random workstations.  (To avoid having multiple RE instances running at the same time, actively running plans on the same hardware.)  By team agreement, we chose one, specific workstation.  The code checks for that.  This additional configuration must be known and checked before `start-re-manager` is run.

rodolakis, 2024-10-17:
> QS Host and QS clients are separate processes.
> 
> * QS Host process is started from a workstation with access to:
>   
>   * the `qs-config.yml` file
>   * a `redis` service
>   * the instrument package with the RE, devices, and plans

Right, I am perfectly good with that part. But I am still unclear about how, in practice, do you achieve that:

> * Any QS client is started from a workstation with network access to the QS Host process

Do you need  that (QS client) workstation to have the package installed with a different set of settings for `qs-config.yml`? Or a lighter version of the package (since it does not need to know about the RE, devices and plans, which are handled by the host)? How do you start the Queuemonitor from that workstation?

prjemian, 2024-10-17:
QS clients will never use the `qs-config.yml` file.

prjemian, 2024-10-17:
> Or a lighter version

The packages required depend on which client program is to be run.  For example, the standard GUI (`queue-monitor`) comes from the [`bluesky_widgets`](https://blueskyproject.io/bluesky-widgets/) package.

prjemian, 2024-10-17:
To run a client, follow the installation steps for that client.  They don't need the same suite of packages that the server needs.

prjemian, 2024-10-17:
Q: Does a QS client need to know about the redis server or the `qs-config.yml` file?
A: No.  It only needs to know how to communicate with the QS Host process.

prjemian, 2025-03-10:
This is more a documentation issue and can be resolved after 1.0.0 release.

prjemian, 2025-09-26:
Time to revisit this now, since one beamline asked today about how to connect a QS client on workstation B with a QS server on workstation A.

> I launched qserver / redis on a machine named `A` and I will like to connect to the qserver on a different machine on the same subnet, and it seems that I don't have zmq connection.