BCDA-APS/BITS #4: README: suggestions for additional instructions

rodolakis, 2024-10-17:
1. **Basic instructions about redis**: 
- needs to be installed by IT as a service on the QS host, with the `.conf ` file to follow is` /etc/redis/redis.conf` on `otz`
- check `redis` is running on the QS host: `systemctl status redis`

2. ** how to run the QS as a client**

3. **how run/test the queueserver from the queue monitor**

- Connect to the host
- Change tab, open the environment
- Open plan editor, add sim plans to queue
- Start queue
- Check the plans did run successsfully in the console: the simulated plan run so fast that it is hard to tell if they actually ran, need to look at the console - in the other tab. Not trivial for someone who has never used the queueserver.

rodolakis, 2024-10-17:
These instructions will need some translation into our current names.  They were used at the time of this presentation (2024-02-09).  For completeness, I'm copying the subset of slides.  Includes APS Data Management (which is an additional feature).

## general

[image]

[image]

[image]

To watch the QS Host console on your terminal, use this bash command: `qserver-console-monitor`.
Type ^C to exit this.

<details>

## when using data management

[image]

[image]

[image]

[image]

[image]

[image]

</details>

Comment from @prjemian in https://github.com/BCDA-APS/bs_model_instrument/issues/3#issuecomment-2420070677

prjemian, 2024-10-17:
These instructions will need some translation into our current names.  They were used at the time of this presentation (2024-02-09).  For completeness, I'm copying the subset of slides.  Includes APS Data Management (which is an additional feature).

## general

[image]

[image]

[image]

To watch the QS Host console on your terminal, use this bash command: `qserver-console-monitor`.
Type ^C to exit this.

<details>

## when using data management

[image]

[image]

[image]

[image]

[image]

[image]

</details>

prjemian, 2025-03-10:
After 1.0.0 release