BCDA-APS/BITS #5: ipython profile crashes ipython session after installing the package

rodolakis, 2024-10-17:
I ran into the error below when starting my first `ipython` session after installing the package. The startup folder on my default ipython profile had files - it should not have been the case, but I did not any better a few years ago happen. I had to go to `~/.ipython/profile_default/startup/` and delete the content of the folder.

Not sure if this should be a warning in the README.

```python

(bs_model_env) [nefarian ~/bs_model_instrument 20241016-11:57:00] ipython
Python 3.11.10 | packaged by conda-forge | (main, Oct 16 2024, 01:27:36) [GCC 13.3.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.28.0 -- An enhanced Interactive Python. Type '?' for help.
 
 
/home/beams3/RODOLAKIS/.ipython/profile_default/startup/00-PythonPath.py
/home/beams3/RODOLAKIS/.ipython/profile_default/startup/90-StartDir_FR.py
[TerminalIPythonApp] WARNING | Unknown error in handling startup files:
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
File ~/.conda/envs/bs_model_env/lib/python3.11/site-packages/IPython/core/shellapp.py:403, in InteractiveShellApp._exec_file(self, fname, shell_futures)
    399                 self.shell.safe_execfile_ipy(full_filename,
    400                                              shell_futures=shell_futures)
    401             else:
    402                 # default to python, even without extension
--> 403                 self.shell.safe_execfile(full_filename,
    404                                          self.shell.user_ns,
    405                                          shell_futures=shell_futures,
    406                                          raise_exceptions=True)
    407 finally:
    408     sys.argv = save_argv
 
File ~/.conda/envs/bs_model_env/lib/python3.11/site-packages/IPython/core/interactiveshell.py:2932, in InteractiveShell.safe_execfile(self, fname, exit_ignore, raise_exceptions, shell_futures, *where)
   2930 try:
   2931     glob, loc = (where + (None, ))[:2]
-> 2932     py3compat.execfile(
   2933         fname, glob, loc,
   2934         self.compile if shell_futures else None)
   2935 except SystemExit as status:
   2936     # If the call was made with 0 or None exit status (sys.exit(0)
   2937     # or sys.exit() ), don't bother showing a traceback, as both of
   (...)
   2943     # For other exit status, we show the exception unless
   2944     # explicitly silenced, but only in short form.
   2945     if status.code:
 
File ~/.conda/envs/bs_model_env/lib/python3.11/site-packages/IPython/utils/py3compat.py:55, in execfile(fname, glob, loc, compiler)
     53 with open(fname, "rb") as f:
     54     compiler = compiler or compile
---> 55     exec(compiler(f.read(), fname, "exec"), glob, loc)
 
File /home/beams3/RODOLAKIS/.ipython/profile_default/startup/90-StartDir_FR.py:8
      3 import os
      4 #os.chdir("/home/beams22/29IDUSER/Documents/User_Macros")
      5 #os.chdir("/home/beams3/RODOLAKIS/Documents/FR_Macros")
----> 8 os.chdir("/home/beams/RODOLAKIS/bluesky")
     10 print("\nWARNING: You are running as FANNY with default profile set for bluesky_2023_2")
 
FileNotFoundError: [Errno 2] No such file or directory: '/home/beams/RODOLAKIS/bluesky'
 
In [1]: 
```

prjemian, 2024-10-31:
Agree, good for Warnings section of the README.

prjemian, 2025-03-10:
After 1.0.0 release.  It's an IPython problem to remove cruft.  Might be a FAQ.

MDecarabas, 2025-04-22:
@prjemian @ravescovi I want to close this out to clean out the trash. Given the new setup do we still encounter this?

prjemian, 2025-04-22:
Close it.  Refers to an unusual situation.  Problem will be different when it happens again.

closed 2025-04-22 by MDecarabas