BCDA-APS/BITS #22: Mapping of "switch" cases

ravescovi, 2024-11-13:
It would be good to create a list / map of all switch cases (tools that can be switched on/off on the iconfig) like the bluesky magick.
`USE_BLUESKY_MAGICS: True`

And which part of the code this "switch" actually acts on. In principle all switches should only act on "BITS" parts of the code to permit for rapid updating of existing templated packages. We should also make sure to pack together all necessary "extra" parameters for each switch in the iconfig.

prjemian, 2024-11-13:
As mentioned in #21

prjemian, 2025-02-03:
What is `"BITS"`?

ravescovi, 2025-02-04:
Bluesky Instrument Template'S

prjemian, 2025-02-07:
Might we change the name of this repo to `bits`?  IMO, `bs_model_instrument` is awkward to use in discussion.

prjemian, 2025-02-07:
GitHub makes this easy, even providing redirects for users still using the old name.  Up until the (unlikely) time that a new repo with that old name is created.

prjemian, 2025-02-10:
@MDecarabas  and I agreed to change the name right away.

ravescovi, 2025-02-10:
This is awesome. My bad about using BITS before asking. 
I've been using as an achronym on other projects and it sliped my mind. 
I think it is also good the "rename" since we can also create some status like
32ID is fully BITS, 8ID is almost BITS. etc. Once we get out of beta, we can also version control it.

prjemian, 2025-02-10:
Just created a milestone for [v1.0.0 release](https://github.com/BCDA-APS/BITS/milestone/1) and assigned all open & closed issues/PRs to it.  We can pull out any of the open issues on a case-by-case basis if we choose to handle them later.

ravescovi, 2025-02-10:
This is awesome. I will read later today. Thanks Pete.

PR #47 (2025-03-13): MNT: BITS structure cleanup
# Description

Addressing Switch cases. PR is also gonna be used to test Issues: #43 

- Closes: #22 
- Closes: #44 

## Type of change

- Code maintenance/cleanup
closed 2025-03-13 by MDecarabas