BCDA-APS/BITS #29: Map of deployments

ravescovi, 2024-11-25:
I think it would be useful to have something like this figure here with the map of Bluesky/BITS deployments on the APS.
[image]
I think @prjemian also had a table with status and needs for the deployments. That would be very useful to track which version and what is needed to keep the consistency over the repo's.

ravescovi, 2025-03-07:
@prjemian do you know who has this figure for the new APS-U ring? Would be really cool to have one with all beamlines that are using BITS. Also, this kinds of brings me the idea that BITS needs a logo.

prjemian, 2025-03-07:
Unless the ANL rules have changed, we're not allowed logos.

ravescovi, 2025-03-08:
I think you showed a document before with all the beamlines that were using (going to use) bluesky. Is there a new version with BITS? I think this could be a cool thing to add in the documentation of this repo, what you guys think?

prjemian, 2025-03-08:
I want to keep that document internal for planning purposes.  It's in Box.

closed 2025-03-10 by prjemian