BCDA-APS/BITS #43: ophydregistry has an update

prjemian, 2025-02-07:
@canismarko communicated on Teams:

> I just pushed v2.0.0rc0 of the ophyd registry to pypi that implements the new behavior we discussed that allows multiple devices to have the same name. It did not raise any warnings when used with Guarneri and the example (@prjemian) gave .
>  
> It also adds type hints that pass mypy and marks ophyd-registry as a typed package.
> 
> I went with a new major version number since there are two changes that technically aren't backwards compatible:
``Registry.findall()`` now returns a list of devices instead of a generator.
Allowing duplicate names for devices is technically new behavior. I don't think that should cause problems, but to be safe I incremented the major version.
If you could, would you mind seeing if this fixes the issue you originally had with Guarneri issuing spurious warnings, and let me know if there are any other problems?
> 
> https://pypi.org/project/ophyd-registry/2.0.0rc0/

ravescovi, 2025-03-04:
Related bug reports on #44 and  #46.

prjemian, 2025-03-10:
Done

closed 2025-03-10 by prjemian