BCDA-APS/BITS #45: Win11 AttributeError: get_ipython() has no attribute "magic"

prjemian, 2025-03-01:
While researching #44:

Using `ipy = get_ipython()`, the `ipy` object has no attribute `magic`.  Something has changed, at least on Windows.  Probably need to refactor to be more general.  [IPython magics Reference](https://ipython.readthedocs.io/en/stable/api/generated/IPython.core.magic.html)

https://github.com/BCDA-APS/BITS/blob/a48d6ccd9ebe373e5835b7128413de88c6545055/src/instrument/utils/logging_setup.py#L196

prjemian, 2025-03-07:
One thing that has changed recently, ipython 9.0.0 was released on conda-forge 5 days ago:
[image]

Similar for PyPI
[image]

But, the initial problem report pre-dated this release by at least four days.  Suspecting that, while it may be factor in this issue, ipython version is not the root cause of #44.

ravescovi, 2025-03-08:
This was fixed in main.

closed 2025-03-08 by ravescovi