BCDA-APS/BITS #46: Calling RE(make_devices()) twice raises a lot of errors.

ravescovi, 2025-03-04:
**Describe the bug**
Running RE(make_devices()) twice raises a bunch of errors.

**To Reproduce**
Steps to reproduce the behavior:
On a clear installation on ubuntu 24.04
`pip install -e .`
`from instrument.startup import *`
`RE(make_devices())`

**Expected behavior**
I think there needs to catch that is already in place?

**Error Code**
```
(bs2) ravescovi@ripchip:~/workspace$ python
Python 3.11.11 (main, Dec 11 2024, 16:28:39) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from instrument.startup import *
I Tue-15:39:10.806: **************************************** Bluesky Startup Initialized
W Tue-15:39:12.846: APS DM setup file does not exist: '/home/dm/etc/dm.setup.sh'
I Tue-15:39:14.148: Databroker catalog: temp
I Tue-15:39:14.247: RunEngine metadata saved in directory: .re_md_dict.yml
I Tue-15:39:14.954: using ophyd control layer: 'pyepics'
I Tue-15:39:15.596: SPEC data file: /home/ravescovi/workspace/20250304-153915.dat
>>> RE(make_devices())
Exception in thread Thread-4 (internal):
Traceback (most recent call last):
  File "/home/ravescovi/miniconda3/envs/bs2/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/home/ravescovi/miniconda3/envs/bs2/lib/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ravescovi/miniconda3/envs/bs2/lib/python3.11/site-packages/apstools/plans/run_blocking_function_plan.py", line 67, in internal
    result = function(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ravescovi/workspace/BITS/src/instrument/utils/make_devices_yaml.py", line 95, in _loader
    setattr(main_namespace, label, oregistry[label])
                                   ~~~~~~~~~^^^^^^^
  File "/home/ravescovi/miniconda3/envs/bs2/lib/python3.11/site-packages/ophydregistry/registry.py", line 172, in __getitem__
    return self.find(key)
           ^^^^^^^^^^^^^^
  File "/home/ravescovi/miniconda3/envs/bs2/lib/python3.11/site-packages/ophydregistry/registry.py", line 362, in find
    raise MultipleComponentsFound(
ophydregistry.exceptions.MultipleComponentsFound: "Found 2 components matching query [any_of=shutter, label=None, name=None]. Consider using ``findall()``. [SimulatedApsPssShutterWithStatus(prefix='', name='shutter', read_attrs=['busy', 'open_signal', 'close_signal', 'pss_state'], configuration_attrs=[]), SimulatedApsPssShutterWithStatus(prefix='', name='shutter', read_attrs=['busy', 'open_signal', 'close_signal', 'pss_state'], configuration_attrs=[])]"
()
>>> 

```
**Environment (please complete the following information):**
 - OS: Linux Ubuntu 24.04

prjemian, 2025-03-04:
Should be possible to run `RE(make_devices())` any number of times in the same session.  There is a [setting](https://github.com/prjemian/prj_BITS/blob/5444c831b4cf82899f8aa7351f5e38f52aa371d6/src/instrument/startup.py#L63C5-L63C38) to not warn.  `oregistry.warn_duplicates = False`

canismarko, 2025-03-05:
If you mean for ``make_devices()`` to be indempotent (ish), maybe there should be a call to [``oregistry.clear()``](https://github.com/spc-group/ophyd-registry?tab=readme-ov-file#removing-devices) somewhere at the start of ``make_devices()``?

Otherwise, I don't know of a way to know that you're adding a duplicate of an existing device, since equality and identity both fail here:

```python
In [1]: from ophyd_async.epics.motor import Motor

In [2]: m1_original = Motor(name="m1", prefix="255id:m1")

In [3]: m1_duplicate = Motor(name="m1", prefix="255id:m1")

In [4]: m1_original is m1_duplicate
Out[4]: False

In [5]: m1_original == m1_duplicate
Out[5]: False
```

I think the warnings are serving their intended purpose. When you run ``make_devices()`` a second time, you are genuinely creating a new object for each device, and trying to add it to the registry. If you ignore the warnings, you will then have two versions of every device and sorting them out will be a chore.

The Haven instrument loader (specific to the SPC group) [calls ``registry.clear()``](https://github.com/spc-group/haven/blob/19c0007faf7e79f01926c86bc2369792e401c40f/src/haven/instrument.py#L59C1-L60C33) at the start of loading devices.

The ``guarneri.Instrument.load()`` does not include ``registry.clear()``, since I think that would make it impossible to load devices from more than one file. Instead, Guarneri leaves it up to the clients to decide when and whether to clear the registry.

ravescovi, 2025-03-05:
To be clear. The reason why I did `RE(make_devices())` twice was by pure mistake. I would expect some clean warnings "deviceX already exists. Skipping" for literally every single device (because they are ALL duplicate) not an error inside a thread.

Now. About clearing the devices, I agree this looks like an elegant solution.

prjemian, 2025-03-05:
## Use case: revising the `devices.yml` file

When editing the `devices.yml` file, it is useful to load the current edit, replacing all objects loaded previously. The `make_devices()` plan helps this process by avoiding the need to restart the session, just to try out the revision.

In this case, to avoid complaints about duplicate entries (from the previous round), clear the registry.  Yet, you *want* to be informed of duplicate entries (such as two different PVs mapped to the same object name).  Likely an oversight due to copy/paste error.

## Use case: queueserver

It is conceivable that `make_devices()` could be extended to take a file argument.  From any session that can run plans (such as queueserver), a new and different set of objects could be installed (with optional keyword argument to clear all existing objects, or not) for the next plans to be run.

canismarko, 2025-03-05:
> with optional keyword argument to clear all existing objects

This is how Haven handles it (though we don't have a Bluesky plan for this at the moment).

prjemian, 2025-03-05:
Such change would refactor the existing code: https://github.com/BCDA-APS/BITS/blob/59432649eb49498dc8c503156302606f470e94b6/src/instrument/utils/make_devices_yaml.py#L56-L63

so that it loads a single, named file (perhaps default to `devices.yml`) with other kwargs as mentioned, including the existing `pause`. https://github.com/BCDA-APS/BITS/blob/59432649eb49498dc8c503156302606f470e94b6/src/instrument/utils/make_devices_yaml.py#L65-L70

Such a change would enable addition files YAML files, similar to `devices.yml`.  I know this pattern would benefit several APS beamlines.  MPE group, for example.  SAXS, for another.

prjemian, 2025-03-05:
In such a case, we should add an identifier to these YAML files so `make_device()` can reject acting on input from any valid YAML file that is not intended for this purpose (such as `iconfig.yml`).

prjemian, 2025-03-05:
Thinking more... such as from https://github.com/BCDA-APS/BITS/blob/59432649eb49498dc8c503156302606f470e94b6/src/instrument/configs/devices.yml#L1-L5

to gather all devices under a `guarneri-devices` key.

```yaml
# Guarneri-style device YAML configuration

guarneri_devices:
    apstools.devices.SimulatedApsPssShutterWithStatus:
    - name: shutter
      labels: ["shutters"]
```

*Any* YAML file with this specific key could be used to provide configuration for `make_devices()`.
This leaves root of that file open to other keys which may be useful in some other context.  I'm thinking of pre-configuration for certain devices (like area detectors) which are created by this YAML.

prjemian, 2025-03-05:
Alternative name of this key: `bits-devices` (local repo branding, leaves open how the devices are actually created).

prjemian, 2025-03-05:
Example (from USAXS) of additional configuration steps:

```py
    # examples: make shortcuts to specific channels assigned in EPICS

    timebase = scaler1.channels.chan01.s
    I0 = scaler1.channels.chan02.s
    scint = scaler1.channels.chan03.s
    diode = scaler1.channels.chan04.s
    I000 = scaler1.channels.chan05.s
    I00 = scaler1.channels.chan06.s
```

commit 9cd47951 (2025-03-23): FIX #46
commit 9f05191e (2025-03-23): FIX #46 Also remove devices from  __main__
closed 2025-03-24 by prjemian