BCDA-APS/BITS #53: Changes proposed by template-sync workflow were already reviewed and dismissed in previous PR.

prjemian, 2025-03-09:
**Describe the bug**
Working with a repo created from this template, seeing the same changes proposed by the `template-sync` workflow were already reviewed and dismissed in previous PR.

**To Reproduce**
1. Use this template to create a new repo.
2. Configure for the local instrument's controls, affecting various files.
3. Run the `template_sync` workflow.
4. Revise the PR it creates.
  - Probably need to recover some of the changes it proposes so that local content is not removed.
6. Merge the PR.
7. Run the `template_sync` workflow again.
8. See changes proposed *again* that would remove local content.

**Expected behavior**
Local content (once reviewed in previous PR) should not be removed by new PR from `template_sync` workflow.

**Additional context**

See these PRs:

- https://github.com/prjemian/prj_BITS/pull/6
- https://github.com/prjemian/prj_BITS/pull/5
- https://github.com/prjemian/prj_BITS/pull/4
- https://github.com/prjemian/prj_BITS/pull/3

prjemian, 2025-03-09:
This type of thrashing indicates that we have a problem separating BITS content from beamline content.

One simple indicator is: https://github.com/BCDA-APS/BITS/blob/4ef001e3f8fc1d479ee4336a9fba3a5cc4b8e590/src/instrument/configs/iconfig.yml#L9

Once a beamline localizes from this template, this value will be changed to the local catalog name.  This file, and assumptions in the associated unit tests in `test_general.py` are affected.

Another file expected to be changed by the beamline is `devices.yml`.  To escape continued reversion to BITS version, my [instrument](https://github.com/prjemian/prj_BITS) added a [new YAML file](https://github.com/prjemian/prj_BITS/blob/main/src/instrument/configs/devices_prj.yml) for its devices.

prjemian, 2025-03-09:
This affects PR #38.

prjemian, 2025-03-10:
See https://github.com/AndreasAugustin/actions-template-sync/issues/588

ravescovi, 2025-03-10:
Some of this behaviour of creating new files can be added to the init repo.

ravescovi, 2025-03-20:
Obselete since moving to pypi

closed 2025-03-20 by ravescovi