BCDA-APS/BITS #62: Re-arrange

prjemian, 2025-03-13:
Looking at #58.  I'm expecting more from the `init_repo.sh` installer.  Resulting in an initial instruction such as:

```bash
ipython -c "from i12id_saxs.startup import *"
```

where `i12idsaxs` is the local "instrument".  `BITS` is not the _main_ part of this, it's just called to set up the instrument in a way that the beamline can understand and the bluesky team can support.

Here's the installed directory under `src/bits`:

```bash
(local_bits) zorinvm@zorin22:~/.../BCDA-APS/local_bits$ ll src/bits/
total 40K
drwxrwxr-x 8 zorinvm zorinvm 4.0K Mar 13 11:19 ./
drwxrwxr-x 3 zorinvm zorinvm 4.0K Mar 13 09:47 ../
drwxrwxr-x 3 zorinvm zorinvm 4.0K Mar 13 11:22 callbacks/
drwxrwxr-x 3 zorinvm zorinvm 4.0K Mar 13 11:22 core/
drwxrwxr-x 6 zorinvm zorinvm 4.0K Mar 13 11:22 demo_instrument/
-rw-rw-r-- 1 zorinvm zorinvm  432 Mar 13 09:47 __init__.py
drwxrwxr-x 2 zorinvm zorinvm 4.0K Mar 13 11:19 __pycache__/
drwxrwxr-x 2 zorinvm zorinvm 4.0K Mar 13 09:47 tests/
drwxrwxr-x 3 zorinvm zorinvm 4.0K Mar 13 11:19 utils/
-rw-rw-r-- 1 zorinvm zorinvm  542 Mar 13 11:12 _version.py
```

We're still upside-down on the way the installed directory should look.  The priority should be on the local instrument.  All its resources should be in the "top" directory.  Any `bits` directories should be one level down.  This is social engineering to reduce the risk of folks customizing the wrong files.

Refactor like this:

```bash
src/
   i12idsaxs/  # the `$repo` term in `init_repo.sh`
      bits/  # this is the directory we can update
      callbacks/  # empty
      config/  # starting files
      devices/  # empty
      plans/  # empty
      utils/  # empty
```

prjemian, 2025-03-13:
Basic aim of BITS: set it up and then get out of the way.  We're doing that first part.  The second part is what this issue describes.

prjemian, 2025-03-13:
Sounds good if we make the packaging/installation steps work.

prjemian, 2025-03-13:
It's that `bits_instrument` that will be renamed during `init_repo.sh` (or maybe `init_repo.py`)?

ravescovi, 2025-03-13:
I think bits and demo_instrument will not be renamed at all. Just the update the README and documents.

prjemian, 2025-03-13:
As an instrument template, we are doing two things.

- standard way to layout an instrument  (the `demo_instrument` part)
- standard way to start a bluesky session (the `bits` part)

At some point, I see that second part as a different package entirely.

prjemian, 2025-03-13:
`demo_instrument ` -> `instrument` or even shorter synonym, not longer

ravescovi, 2025-03-13:
I just worry about the complexity of that. at Gladier we have 

- Gladier >> Main package
- Gladier-tools >> collection of usefull tools
- Gladier-client-template >> The equivalent of our demo_instrument template
- and many others. 
I think it became a little overwhellming to maintain all those up to date with each other.

On this proposal we would have
- BITS >> main package (could be published to pypi and carry all documentation)
- BITS_template >> template package to be cloned and carry all BITS functionality.
- BITS_training >> main documentation package (could probably go into BITS then)

ravescovi, 2025-03-13:
I think I see what you are saying. Right now.
When I isolate `src/bits` and `src/new_instrument`, I am basically isolating `BITS` and `BITS_template`. So, the sync behaviour itself would be much simplified if those lived in different repos.

prjemian, 2025-03-13:
For a future release, let's get v1.0 out the door.

ravescovi, 2025-03-20:
I think this is done.

closed 2025-03-20 by ravescovi