BCDA-APS/BITS #66: BITS on pypi

ravescovi, 2025-03-17:
It is starting to get clear to me we need to have `BITS` as a pypi package instead of a local package. This will simplify a lot of the user interactions. 
@prjemian was it you that set APS-tools as pypi?
https://github.com/BCDA-APS/apstools/blob/main/.github/workflows/pypi.yml ? 

Should we bring this same idea over to bits?

prjemian, 2025-03-17:
Can do.

ravescovi, 2025-03-17:
What do you think about doing all the sync through pypi?
Is there any reason for BITS to be local (apart from the "instrument"
packages)?

prjemian, 2025-03-17:
As it is now, right?  Let's start that way.

ravescovi, 2025-03-17:
Just so we agree on the same thing.
Right now we will distribute the local version of BITS.
Once we isolate `bits-core` from `bits-instrument` we can publish at pypi
and trivially migrate everyone into the pypi (non-local) dev of bits.

prjemian, 2025-03-17:
So publish to PyPI is planned but needed for 1.0.0.  It's a later milestone (say 1.1).

ravescovi, 2025-03-18:
Do we still need "sync_repo" if we are going to do the syncing through pypi?

prjemian, 2025-03-18:
Maybe not.  Won't hurt to leave it in for now.

ravescovi, 2025-03-19:
BITS have been release to pypi as apsbits
`pip install apsbits` should install v1.0.0-rc2

closed 2025-03-20 by ravescovi