BCDA-APS/BITS #77: How to guarantee BITS and DEMO-BITS are in sync?

ravescovi, 2025-03-20:
**Describe the question**
Is there a way of linking the demo_instrument folder from apsbits into demo-bits?

Probably some kind of GHA.

prjemian, 2025-03-21:
> I think this is an over engineered solution, and perhaps the problem comes
> from the DEMO-BITS name in and of itself. The way I see that package is
> very much as a template from where to build instruments. It should not come
> with a defecto demo_instrument that gets then renamed when templated.
> Instead it should be an empty template where the first step, after forking,
> is the user creating a new instrument using the create-instrument function.
> This way there we no longer have to worry about keeping 2 identical pieces
> of code in sync from different repositories

Agreed. The demo instrument looks odd now in a templated repo.

ravescovi, 2025-03-21:
> Agreed. The demo instrument looks odd now in a templated repo. <br> <br>
> &gt; I think this is an over engineered solution, and perhaps the problem
> comes <br> &gt; from the DEMO-BITS name in and of itself. The way I see
> that package is <br> &gt; very much as a template from where to build
> instruments. It should not come <br> &gt; with a defecto demo_instrument
> that gets then renamed when templated. <br> &gt; Instead it should be an
> empty template where the first step, after forking, <br> &gt; is the user
> creating a new instrument using the create-instrument function. <br> &gt;
> <br> &gt; of code in sync from different repositories <br> &gt; <br> &gt; —
> or <br> &gt; unsubscribe <br> &gt; &lt;
> <br> &gt; <br> &gt; I think this is an over engineered solution, and
> perhaps the problem comes <br> &gt; from the DEMO-BITS name in and of
> itself. The way I see that package is <br> &gt; very much as a template
> from where to build instruments. It should not come <br> &gt; with a
> defecto demo_instrument that gets then renamed when templated. <br> &gt;
> <br> &gt; is the user creating a new instrument using the create-instrument
> function. <br> &gt; This way there we no longer have to worry about keeping
> 2 identical pieces <br> &gt; of code in sync from different repositories
> GitHub <br> &gt; &lt;
> this thread.Message <br> &gt; ID: ***@***.***&gt; <br> &gt; <br>

Then we need to add a check behavior so people don't clone a package and
try to run with nothing inside. I think this is related to the how to ci
discussion too.

MDecarabas, 2025-03-21:
Do you not think the README instructions are enough to point people towards how to use the template?

ravescovi, 2025-03-21:
I am not sur about it. When we open a package called demo and it is an empty python package. I find it a bit weird, where do I go to see how a reference looks like? I need to clone the package and do a bunch of commands?

ravescovi, 2025-03-21:
Is that the same solution for qserver? It does not exist without having an instrument folder inside.
I don't see what is over-engineered, we got rid of the sync behaviour and I created a demo place to showcase how an instrument without `src/apsbits` could be installed. 

I don't have strong opinions either way but if we are providing an empty repository, then it is almost better to just be a set of instructions. It is NOT a template then.

ravescovi, 2025-03-21:
Maybe `src/apsbits/demo_instrument` could go back to `src/demo_instrument and that is the reference we need to look for. but that has some strong software implications on `apsbits` and I would probably be against it.

ravescovi, 2025-03-21:
We could move the qserver files into `configs` and expand the behaviour of `create-bits` to create the shell scripts necessary for qserver/etc. This solves the qserver on instrument.

ravescovi, 2025-03-21:
Mark has the qserver on the package level:
https://github.com/spc-group/haven/tree/main/src
I do like this solution.

maybe we create demo_qserver to serve as template for that and then we add --qserver to the create-bits function

I like this solution but I still find weird to have an "empty" template as reference point

ravescovi, 2025-03-21:
See #80

PR #80 (2025-03-21): Add qs to create-bits.
# Description

This adds the creation of all qserver related files when the user starts a new-bits repository
It also removes the "destination" since all bits packages should be generated inside src.

Closes #77
closed 2025-03-21 by ravescovi