BCDA-APS/BITS #88: Startup hangs indefinitely (need 'cat' object)

prjemian, 2025-03-23:
As part of my local instrument startup I run these plans: `make_devices()`, then `setup_devices()` (a local plan).  In the course of `setup_devices()`, it is necessary to have the `cat` object created here: https://github.com/BCDA-APS/BITS/blob/2a0b2f00291b13479030bf7f5808009d7d675a77/src/apsbits/demo_instrument/startup.py#L37

Since `startup` has not finished when this plan is called, there is an infinite loop (cannot import from uninitialized module or something similar) which is not caught by the `RE`.

While `startup` is running, it is not available from the `__main__` namespace.

The `cat` object is created by `BITS.core.catalog_init.init_catalog()`.  That method could be called but it returns a different `cat` object than is known to `startup`.

How can `cat` be imported and used during startup with this design?

prjemian, 2025-03-23:
[Workaround](https://github.com/prjemian/gpBits/blob/31f03b8dbffec61ef9a3ed14778ed4759e768402/src/instrument/plans/setup_gp.py#L214-L218):

```py
    from apsbits.core.catalog_init import init_catalog
    from apsbits.utils.config_loaders import get_config

    cat = init_catalog(get_config()
    # from ..startup import cat  # FIXME: How to import cat from startup or BITS?
```

prjemian, 2025-03-24:
Kickin this down the road, in case a new idea emerges.

MDecarabas, 2025-04-22:
@prjemian, is this still a thing?

prjemian, 2025-04-22:
We implemented a workaround.  I'm closing it.

closed 2025-04-22 by prjemian