BCDA-APS/BITS #115: Change coming in bluesky-queueserver package

prjemian, 2025-04-30:
With this [change](https://github.com/bluesky/bluesky-queueserver/pull/322) in the bluesky-queueserver code, we'll need to comment out (or remove) this section: https://github.com/BCDA-APS/BITS/blob/ed3e521f795769ce0c377fdfc1c998bff574269a/src/apsbits/demo_qserver/qs-config.yml#L33-L40

Will apply to the next release after the current [v.0.0.21](https://github.com/bluesky/bluesky-queueserver/releases/tag/v0.0.21).

prjemian, 2025-04-30:
Once there is a new release, we can make this change and add a pin in the requirements.  Existing BITS repositories would need this correction to use the new release of bluesky-queueserver.

ravescovi, 2025-04-30:
Sounds good. Almost an opportunity to delete the whole qserver folder and
make it into single exec script.

prjemian, 2025-04-30:
> delete the whole qserver folder and make it into single exec script

Might be the case.  Proof of that idea is contingent on successful definition of the file path described here:
https://github.com/BCDA-APS/BITS/blob/ed3e521f795769ce0c377fdfc1c998bff574269a/src/apsbits/demo_qserver/qs-config.yml#L45-L46

This configuration avoided a need to supply an absolute path to a directory for these two files.

prjemian, 2025-06-25:
Merged and Released:

- https://github.com/bluesky/bluesky-queueserver/pull/322
- https://github.com/bluesky/bluesky-queueserver/releases/tag/v0.0.22

closed 2025-06-25 by prjemian