BCDA-APS/BITS #124: 12id B Saxs convert to BITS

MDecarabas, 2025-05-02:
Convert 12id B saxs to BITS

prjemian, 2025-05-05:
Start a new BITS repo for 12IDB (or E?) in BCDA-APS organization.  Old, initial setups are here but never used.
https://github.com/APS-12ID

prjemian, 2025-05-07:
Sent Teams message to @byeongdu & @jilavsky

Is there any content from the [previous setups](https://github.com/APS-12ID) which must be preserved?

prjemian, 2025-05-07:
Should a single repository be created for both SAXS stations (12-ID-B & 12-ID-E) or separate ones?  These will be separate from the USAXS [repo](https://github.com/BCDA-APS/usaxs-bits).

With a common repo, it is possible to have multiple, separate instruments that share a common library of devices, plans, or other code.

ravescovi, 2025-05-07:
I vote for:

src/common_things/ devices | utils (remove configs/plans and startup)
src/instrument_1  / .... (bits standard)
src/instrument_2 / .... ( bits standard)

In some sense, each instrument is representing a physical beamline location.

Then inside each instrument we can have different startup.py for different
modes/acquisitions/etc.

prjemian, 2025-05-07:
Shorter: `src/common_things/` --> `src/common/`

Agree it does not need `configs/`, `plans/`, and `startup.py`.  Anything else that would never be used from `src/common/`?

prjemian, 2025-05-07:
@Byeongdu commented  on Teams:

> Good to share the same repo for B and E. I believe instruments config will be on each beamline's computer, so will not be mixed up.

> Is there a way to send command to a Bluesky session (running Bluesky in server mode)?

I answered:

> Yes, when running the Bluesky queueserver.  There may be other ways to enable remote interaction with an IPython session.  I've seen such comments in documentation (in the queueserver docs, of all places) but never explored that myself.

ravescovi, 2025-05-07:
The danger of using the name "common" is that this will be installed as a
python package. Probably better to have a full name like "usaxs_common" or
"isn_common"
Apart from that. I think you got it right, remove configs, plans, startups,
callbacks (if not used), suspenders (if not used).

prjemian, 2025-05-07:
12ID SAXS and other groups will need this feature.  Might be good for us to standardize the name of `common_things` so it is easy to use, document, install, and support.  And fold such changes into the BITS repo.

prjemian, 2025-05-07:
https://github.com/BCDA-APS/12id-bits has been created

prjemian, 2025-05-07:
Need to add 12ID staff to this.

prjemian, 2025-05-07:
Have not started any configuration yet, just created the repo.