BCDA-APS/BITS #130: Document how to transfer data from catalog to external location, weather it be a file, a db, or a service like dm

MDecarabas, 2025-05-05:


prjemian, 2025-05-07:
This [document](https://bcda-aps.github.io/bluesky_training/example/_export_bluesky_data.html) describes some various ways to export data.  It was written a few years ago.  Likely other methods have been developed since.

prjemian, 2025-05-07:
Export to CSV: https://bcda-aps.github.io/bluesky_training/example/_export_to_csv.html

prjemian, 2025-05-07:
Write a SPEC file: https://bcda-aps.github.io/bluesky_training/example/_specwriter.html

This could be updated to use the newer `SpecWriterCallback2`.

prjemian, 2025-05-07:
Not to an external location...

Access data *after* the measurement: https://bcda-aps.github.io/bluesky_training/howto/_after_measurement.html

prjemian, 2025-05-07:
Python code to write NeXus & SPEC files (lifted from a standalone application):

```py
    parms = user_parameters()  # via 'argparse'

    # ...

    if len(parms.nexus_file.strip()) > 0:
        nxwriter = NXWriter()
        nxwriter.warn_on_missing_content = False
        nxwriter.file_name = parms.nexus_file.strip()
        RE.subscribe(nxwriter.receiver)
        logger.log(REPORTING_BASIC, "Writing NeXus file: %r", nxwriter.file_name)

    if len(parms.spec_file.strip()) > 0:
        specwriter = SpecWriterCallback2()
        specwriter.newfile(parms.spec_file.strip())
        RE.subscribe(specwriter.receiver)
        logger.log(REPORTING_BASIC, "Writing SPEC file: %r", specwriter.spec_filename)
```

prjemian, 2025-05-07:
On a related topic: [How to Search for Bluesky Data?](https://bcda-aps.github.io/apstools/latest/examples/ho_search_databroker.html)

canismarko, 2025-05-07:
Maybe relevant: https://github.com/BCDA-APS/BITS/issues/131#issuecomment-2852564718

prjemian, 2025-06-02:
This is not a question for BITS, which is an instrument template.