BCDA-APS/BITS #131: finish data acquisition BITS framework by providing a way to extract data generated by a run

MDecarabas, 2025-05-05:
- [ ] Files
- [ ] DM is considered Files
- [ ] Databases

canismarko, 2025-05-05:
In case its helpful, here's how haven is attempting to address this need. Still very much a work in progress, but it does what it says on the tin. Don't have the DM part ready yet, though.

https://github.com/spc-group/haven/pull/420 (nice)

MDecarabas, 2025-05-07:
@canismarko, this is really cool. Would love to pick your brain about it at some point

prjemian, 2025-05-07:
DM does not create files, it manages them.

MDecarabas, 2025-05-07:
yes, hence why the process will overall be the same as files

prjemian, 2025-05-07:
- https://github.com/BCDA-APS/BITS/issues/130#issuecomment-2859039097

> Access data *after* the measurement: https://bcda-aps.github.io/bluesky_training/howto/_after_measurement.html

- CSV - https://github.com/BCDA-APS/BITS/issues/130#issuecomment-2859033137
- NeXus - https://github.com/BCDA-APS/BITS/issues/130#issuecomment-2859051899
- SPEC - https://github.com/BCDA-APS/BITS/issues/130#issuecomment-2859035860

prjemian, 2025-05-07:
DM does not contribute to this issue: ... _extract data generated by a run_

prjemian, 2025-05-07:
On a related topic: [How to Search for Bluesky Data?](https://bcda-aps.github.io/apstools/latest/examples/ho_search_databroker.html)